svalouch/python-rctclient #1: Thank you

ThomasSchoeller, 2020-10-26:
I like to say thank you for this great project.

I have one question.

Is it possible with the cli client to read the data logger values?
I like to generate/automate the tax reporting.

svalouch, 2020-10-28:
Hi, thank you, but I merely ported it to Python 3 and packaged it (see docs for history) :)

Basically, when installing e.g. using pip, specify `rctclient[cli]` and it'll automatically pull in the `click` module that's needed to get the CLI running. Once that's present, the `rctclient` command should start working (if you're using virtualenv, you can also specify the full path to the file, such as `/path/to/virtualenv_folder/bin/rctclient` so you don't have to activate it each time).

For example, to read the current state of charge, use it like this:
```
rctclient read-value --host 192.168.0.1 --name battery.soc
```

Word of warning: If the app or some other software communicates with the device at the same time, the device may return invalid data. In this case, the CLI will terminate with a non-zero exit code so that your script can react to it and (for example) retry.

The full list of commands is available in the documentation (https://rctclient.readthedocs.io/en/latest/cli.html, along with debug options). You can either use the App to find the names ("login" required) of the values that are of interest (like `battery.soc` above) or take a look at the rendered registry at https://rctclient.readthedocs.io/en/latest/registry.html.

Some fields that are in the app are missing from the registry (haven't been discovered yet) and thus can't be queried, I'm positive you won't need any of them, but if you do I'll try to figure them out. If you need the time series or event data (from the history tab in the app), then this won't really work well with the CLI, as they can't be expressed as a single value, but it should be possible to create CSV output with some additional coding.

Let me know if it works or if you need help!

svalouch, 2021-01-26:
Hi @ThomasSchoeller, do you need more help or did it work out for you?

MidnightNerd, 2021-01-29:
Hi everyone, I guess it's not possible to receive data logger values with rctclient cli as they have more than one entry and the CLI doesn't parse it. If i try to receive rctclient read-value --verbose --host xxx.xxx.xxx.xxx --name energy.e_ac_month i get the error:
Frame complete, but buffer still contains 14 bytes
Received unexpected frame, ID is 0x400F015B, expected 0x10970E9D

svalouch, 2021-01-30:
Hi @MidnightNerd, please don't hijack other issues. I've opened a new one for you: #2

svalouch, 2021-02-02:
Right, I misunderstood @MidnightNerd, and I hadn't put any thought about the data that was required in the scenario @ThomasSchoeller was describing, sorry for that. So I overhauled the time series collection script (and renamed it `timeseries2csv.py` from `histogram2csv.py`), it can now query all the series and supports going back a set amount of time based on the resolution of the data. That means that one can specify `--resolution year` to query the per-year time series and use `--count <n>` to select how much time (how many years in this case) to go back. Example:
```
$ tools/timeseries2csv.py --host 192.168.19.13 -r year -o - -c 2 0
Requesting egrid_feed
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting eb
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting ea
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting eext
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting eac
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting egrid_load
	timestamp: 2021-01-02 23:59:59
	Reached limit
Requesting eload
	timestamp: 2021-01-02 23:59:59
	Reached limit
timestamp,egrid_feed,eb,ea,eext,eac,egrid_load,eload
2020-01-02T00:59:59+01:00,-649789.625,476738.4375,463455.53125,0.0,873705.75,4077.275146484375,228041.484375
2021-01-02T00:59:59+01:00,-2019.989990234375,93483.265625,83451.140625,0.0,150273.765625,1286150.875,1434396.375
```
There are some values in there that make no sense, e.g. `egrid_load` being 4k for last year is wrong, but it is identical to what the app displays and I've double-checked that it is indeed what is sent from the device.

Also, while at it, I expanded the CLI, it can now do simple time series queries. They are rather limited:
- they always use the current time stamp to start
- they just return whatever the device responds with (the amount of data greatly varies, especially with the "minute" resolution, and may even be empty).
- 
For example, querying `logger.year_eac_log_ts`:
```
$ rctclient read-value --host 192.168.0.1 --name logger.year_eac_log_ts
2000-01-02T00:59:59=0.0000, 2020-01-02T00:59:59=873705.7500, 2021-01-02T00:59:59=150273.7656
```

The values are printed as they are received, so the first one is the sentinel value the device seems to use for denoting that there may never be more data, the other two values are for last year and the current year, cut off after 4 decimals.

# Note
The script requires a new version of the module, so be sure to get the current HEAD revision, it won't work with the version released on PyPI.

closed 2023-10-05 by svalouch