svalouch/python-rctclient #2: Receive data logger values

svalouch, 2021-01-30:
@MidnightNerd asked in #1:

> Hi everyone, I guess it's not possible to receive data logger values with rctclient cli as they have more than one entry and the CLI doesn't parse it. If i try to receive rctclient read-value --verbose --host xxx.xxx.xxx.xxx --name energy.e_ac_month,
i only get the last entry

svalouch, 2021-01-30:
Before you edited your original message, you mentioned getting a second value in the buffer, I suspect that this was caused by the device getting confused when another client (e.g. the smartphone app) chimed in at the wrong moment, I shall try to come up with a reliable scheme to catch this case in a future revision and implement some sort of retry scheme.

Anyway, the `energy.e_ac_month` is a single cumulative value, the total amount of energy for the current month (so far) and resets at the beginning of a new month (provided that the device's clock is set correctly). These values can be viewed in the (Android) app in the `Device`-tab, navigate to `Measured Values` → `Energy` → Select the range, such as `Month`.

From your question, I guess you're interested in the historical data (Android app: `History`-tab → Graph display). These are in the [logger](https://rctclient.readthedocs.io/en/latest/registry.html#logger) section and querying them is a quite complicated process compared to getting a single value, thus there is no support for that in the `rctclient read-value` command. There is, however, a separate script [histogram2csv.py](https://github.com/svalouch/python-rctclient/blob/main/tools/histogram2csv.py) ([documentation](https://rctclient.readthedocs.io/en/latest/tools.html#histogram2csv-py)) that queries a days worth of histogram data and stores it as CSV that can then be imported, e.g. into a spreadsheet application or a time series database (see `csv2influxdb.py` in the same folder).

The `histogram2csv.py`-script is rather inflexible, it always queries an entire day worth of data at the finest granularity with each invocation and assumes the default sampling rate of 5 minutes (the default "Day" setting in the apps histogram section, which can store the past 90-ish days), which is enough for my own use-cases: I'm pushing the data into a time series database and it calculates monthly statistics for me. If time permits, I may rewrite it into something more elaborate and flexible in the future. Let me know if it's usable for you in its current form, else I can try to whip up something for the monthly values.

MidnightNerd, 2021-01-30:
Your absolutely right, i forgot to disable the app.
But what i wanted to mention is that the poster need these logger-values to generate tax reporting.
This is as you mentioned not possible with rctlient cli.
I guess Self consume (logger.year_eac_log_ts-logger.year_egrid_feed_log_ts) is needed for tax reporting.

svalouch, 2021-02-02:
Alright, you're absolutely correct, sorry for misunderstanding. I hadn't put much thought into the kind of values one would need for that scenario. So, I ended up completely overhauling some parts as mentioned in #1.

closed 2021-02-02 by svalouch