svalouch/python-rctclient #5: please check if you want to change the documentation

UweRieping, 2021-02-18:
today I was playing around with rctclient and noticed the following in the [documentation](https://rctclient.readthedocs.io/en/latest/cli.html) 

Examples:

rctclient read-value –name temperature.sink_temp_power_reduction
rctclient read-value –id 0x90B53336

Please check if you want to change it in: 

rctclient read-value --host 192.168.0.1 --name temperature.sink_temp_power_reduction
rctclient read-value --host 192.168.0.1 --id 0x90B53336

I have submitted it here because I don't know where to place it elsewhere.

svalouch, 2021-02-18:
Oh wow, good catch! Don't worry, this is the correct place :)

Two things are wrong in the documentation: Firstly, it looks like like docutils/Sphinx is trying to be smart here and replaces two dashes with one, which is typographically pleasing but clearly wrong here. The page in question is generated directly from the docstring in [cli.py#L127](https://github.com/svalouch/python-rctclient/blob/v0.0.2/src/rctclient/cli.py#L127) which also powers the `--help` output to the terminal where it's rendered correctly, so I can't do much in that place without breaking the terminal help.

```
$ rctclient read-value --help | grep -A3 Examples:
  Examples:

  rctclient read-value --name temperature.sink_temp_power_reduction
  rctclient read-value --id 0x90B53336
```

And secondly, the example is clearly incomplete and does not run in that form. Originally, I omitted address details to keep it concise, but in the end it is always better to have a valid example that's slightly longer than to have a broken one. I'll fix that right away.

svalouch, 2021-02-18:
Actually, fixing the hyphen was way easier than expected, and I disabled the "smartquotes" functionality altogether.
Many thanks for spotting these problems!

closed 2021-02-18 by svalouch