svalouch/python-rctclient #9: Question

theld001, 2021-04-16:
Hallo,

habe einen RCT Wechselrichter DC10 mit Batteriespeicher und lese im Moment über Modbus (Seriell) die Daten.
Teste gerade den RCTclient um noch mehr Werte aus dem Wechselrichter zu bekommen. 

Bei einem Zugriff mit der App um den Namen zu ändern habe ich mir nun die Kommunikation zum Wechselrichter blockiert so das dieser nicht mehr mit der App und dem rctclient komunizieren kann. Ansonsten funktioniert noch alles. Ein Reset des Wechselrichters brachte keine Lösung. Sieht aus als wäre noch die Portkommunikation blockiert da ein Prozess anscheinend nicht abgeschlossen wurde. Dies scheint der gleiche Effekt zu sein wie wenn eine App aktiv ist dann gibt der rctclient auch keine Antwort.

Schon einmal bei der Entwicklung von rctclient einmal einen ähnlichen Effekt gehabt das die Kommunikation blockierte? Falls ja wie das Problem gelößt
Grüße

svalouch, 2021-05-21:
Hi, sorry fuer die spaete Antwort. Bisher haben sich alle Haenger auf die aktuelle TCP-Session beschraenkt, d.h. ein Abbruch via ctrl+c und neu Aufbauen hat das Problem behoben, spaetestens nachdem alle clients (app, rctclient etc.) keine Netzwerkverbindung mehr hatten (via Trennen des Ethernet-Kabels).

Falls das Problem nicht zwischenzeitlich geloesst wurde (wenn ja wuerde mich die Loesung interessieren): Viele Einstellungen merkt sich der Wechselrichter erst dauerhaft, wenn man in der App auf "Flash" drueckt. Wenn das noch nicht passiert ist koennte(!) es helfen, das Geraet ueber seinen Unterbrecher an der Unterseite einige Zeit stromlos zu machen und dann wieder anzuschalten. Ich weiss allerdings nicht ob das alle Settings betrifft und ob das in diesem Fall tatsaechlich die Loesung ist. Im Zweifel den Support kontaktieren.

schlotzz, 2021-05-23:
> Bisher haben sich alle Haenger auf die aktuelle TCP-Session beschraenkt, d.h. ein Abbruch via ctrl+c und neu Aufbauen hat das Problem behoben, spaetestens nachdem alle clients (app, rctclient etc.) keine Netzwerkverbindung mehr hatten (via Trennen des Ethernet-Kabels).

In meiner Implementierung des RCT-Protokolls werden diverse Werte im Intervall von wenigen Sekunden gepollt. Dabei wird die bestehende TCP-Verbindung offengehalten. Dennoch bricht die Kommunikation scheinbar jeden Tag zusammen und die TCP-Verbindung muss erneut geöffnet werden. Daher würde mich auch brennend interessieren, ob hier etwas bekannt ist...

theld001, 2021-05-23:
Hi in meiner Implementierung starte ich immer zyklisch um ca. 50 Werte zu lesen. Auch dabei kommt es immer wieder zu problemen die ich über einen Timeout abgefangen habe. d.h. Ich lese und wenn innerhalb einer maximal Zeit keine Antwort kommt breche ich ab und lese erneut. Dies Probleme habe ich fast immer wenn mir eine andere Verbindung dazwischen funkt.

closed 2021-05-23 by theld001

theld001, 2021-05-23:
Blockierung der Kommunikation "App zum Wechselrichter" wurde mit Hilfe des Errichters und Supports behoben.

svalouch, 2021-05-24:
> Dennoch bricht die Kommunikation scheinbar jeden Tag zusammen und die TCP-Verbindung muss erneut geöffnet werden.

Okay, das hab ich heute morgen unbeabsichtigt nachgestellt, indem ich den Rechner auf dem meine Metrik-Software (rctmon, noch nicht gepushed) laeufft ueber Nacht in Suspend2RAM gefahren habe. Laut Log hat der Netzwerksocket meine Requests angenommen, aber es kam keine Antwort mehr. Leider hab ich nicht dran gedacht Wireshark anzuwerfen; beim naechsten mal dann. Davor lief sie 3 Tage lang ohne Probleme durch. Ich fahre hier uebrigends Firmware `2.3.5198`.

> Blockierung der Kommunikation "App zum Wechselrichter" wurde mit Hilfe des Errichters und Supports behoben.

Interessant, koenntest du das ein bisschen ausfuehren? Klingt nach was dass man in einen FAQ aufnehmen koennte :-)