svalouch/python-rctclient #12: Feature Request: Values for Autarky and Self-consum

leonardo0014, 2021-06-03:
Hello

thanks for the quick help the last time and the current updates. When reading out the values, I noticed that three values are missing for the day, month and year data (see LINK). These are "Autarky [%]", "Self-consum [%]" and "Self-consum [kWh]". Are they that new or do they have to be calculated somehow? In any case, they are not listed in the registry section. If your system is too old for this, I can try to determine the OID on my own with the appropriate instructions.

I would appreciate an answer
-----------------------------------
[1] LINK: https://www.rct-power.com/de/download-bereich-de.html?file=files/Download%20Bereich/Download%20Bereich%20DE/8_Technische%20Dokumentation/TD15_Erl%C3%A4uterung%20Datenk%C3%BCrzel%20Graphik%20in%20Power%20App_Web21V1DE.pdf

svalouch, 2021-06-04:
It looks like these are calculated by the app. Here is how I checked and how you can reproduce it (please do if you can, neither me nor these tools are perfect!):

1. Start a `tcpdump`, e.g. `tcpdump -w inverterp.pcap host 192.168.0.1` if that's the IP of the inverter. The tricky part is to get the dump from a device that can see all traffic to/from the inverter (optimally) or at least the traffic from the client device. I run tcpdump on my router directly so I'm sure that it catches everything.
2. Open the diagram view in the app in "Month" view and check `Autarky [%]`, `Self-consum [%]` and (for reference) `Eext [Wh]`. Wait for the full diagram to be loaded.
3. Close the app and a few seconds later stop tcpdump.
4. Use `tools/read_pcap.py inverter.pcap` to analyze the dump.

I then filtered traffic that the app produces to update the main screen (such as `power_mng.*`, `g_sync.*`, `dc_conv.*`, `battery.*` and `fault*`) and ended up with the following three OIDs in the logger section:

* `logger.day_eext_log_ts`
* `logger.day_egrid_feed_log_ts`
* `logger.day_eload_log_ts`

There may be lines where the tool states `Could not find ID in registry`, but in this test all of them were instances where it tried to extract some useful information after a CRC error, i.e. all of these occurrences were invalid data. As I can't find anything else that it could use I came to the conclusion that it does the calculation in the app. There are a few instances where it is known that the app changes the data it gets from the inverter, e.g. by flipping negative values to positive values, and it makes sense in this case to do the calculation in the app rather than use more space in the inverters limited memory.

UweRieping, 2021-06-07:
There are a few calculated values:
my investigation and calculation is the following:

Edc [Wh]	              = ea + eb
Autarky [%]	      = (100-(egrid_load * 100/elaod))
Self-consum [Wh] = eac - egrid_feed
Self-consum [%]    = (Self-consum * 100 / eac) /// = (eac - egrid_feed) * 100 / eac

leonardo0014, 2021-06-13:
I'm sorry for the late reply, unfortunately I haven't had the chance to try it out yet.

@UweRieping 
Thanks for the answer. Just one more question to understand the answer: What do you mean by "/// ="?

@svalouch 
Thanks for the answer. 
I implemented that right away, just puzzling around the output. Could it be that the read_pcap script has an error? 
All results are "1617321599". 
I've attached the expenses ([export060621_1311-issue.txt](https://github.com/svalouch/python-rctclient/files/6643656/export060621_1311-issue.txt)). Up to line 35 that of the app and from there the output of "tools/read_pcap.py".

Thank you in advance for the answers

UweRieping, 2021-06-14:
"///" is only a logical separator. You can ignore it. 
Self-consum [%] = (Self-consum * 100 / eac) 
Self-consum [%]= (eac - egrid_feed) * 100 / eac

leonardo0014, 2021-06-25:
Thanks all for the Help

closed 2021-06-25 by leonardo0014