svalouch/python-rctclient #15: Feature: write value

MichaelMMS, 2022-02-05:
It would be nice if you could also send write commands to the inverter. That way you could build better automations instead of having to use the rct app manually.

something like:

` rctclient write-value --id 0xXXXXXXXX --value 0.5 `

svalouch, 2022-02-17:
In theory, write-support is very trivial to implement: You basically send a frame with a `WRITE` or `LONG_WRITE` command to an OID and encode the payload according to its entry in the registry.

However, I have not implemented it for a the following (non-conclusive) reasons (which I should probably add to the documentation):

- I know for a fact that certain values must be within bounds, such as maximum or minimum value for an integer, or maximum length for a string. What little I have in regards to documentation does not include this information. Yet the client would need to know, to prevent accidentally causing damage.
- Damage? Well, I have no idea how the device deals with invalid data. If it does not have its own validation and plausibility checks, it may apply invalid values and cause issues in the physical world, e.g. by over-stressing its own components, impacting the power grid or cause damage to electronics within the house. For all I know, it completely relies on the App to prevent invalid data from being sent. But I'm not sure, because:
- The protocol lacks any measure to communicate failure of any sorts. While the device has a rich set of error conditions that can be queried, it does not have a feature to communicate "this value is invalid". I don't know how it will react, it may simply ignore it or send a specific value back that is not part of the documentation, or apply it somehow. What would happen if the trailing `\0` on a string was missing, or if the payload was far larger than what the device supports, for example? Would it run over the intended length of the string into its own adjacent memory, causing it to… do what, crash and reset, or would it become bricked? I have received padded string payloads that contained garbled data after the `\0`, which to me hints at less-than-ideal memory management, but where? On the network interface card or the main processor? Which of them would react, and how? How would you recover from it, or would you need to call the vendor and hope for a replacement?
- Some OIDs are not meant to be written to, yet the device does allow writing to at least some of them and reports the new value in further read-requests. The documentation does not include information on which are writable or supposed to be written to, or which of these are persisted to flash. Yet the client would need to know this to prevent overwriting important information.
- The vendor provides *zero* support for usage of the device by means other than their App or portal. I certainly don't get answers from them, they didn't even send me documentation, and others who tried to implement the protocol in their own applications have reported that while they got access to a -- presumably castrated or even simulated -- device to test against, the support did not answer any of their questions regarding specifics related to the protocol, let alone how the devices react or how one is supposed to react to certain behaviors that the device shows.
- I don't have the means to test anything in that regard. The only thing I can do is extract `WRITE`s issued by the app from a tcpdump and compare it to the output of `make_frame` with it to see if the syntax is correct. But I can only generate a very limited set of these calls, as my device is in active use.

If you want to implement this on your own: Values are by ephemeral by default, meaning their values will be reset when the device reboots. To persist them, you need to do the COM-dance by writing `0x05` to `com_service` and then set it back to `0x00` after a couple of seconds. If this was a success is unknown, one knows when the next reboot happens and the old value shows up. Also, this wears the flash down, these chips typically have a very limited lifetime (anywhere from a few dozen to a couple hundred writes, depending on the actual chip and how it is implemented), so one does not want to write to flash other than absolutely necessary, and the device most likely breaks when the flash is dead. There are certain counters that can be queried that seem to report the number of write-cycles though, but it's not specifically documented and how many it can sustain is unknown.

# TL;DR: No, sorry.

That being said, the client *does* support writing: The timeseries data is queried by `WRITE`ing the current unix timestamp to the device and it answers with a `timestamp: value`-table. That was, by the way, reverse-engineered by staring at hexdumps, as the format is not in the documentation. But all the code is there, just not wired up the way you want.

> PS: […]

Thank you for your kind offer, but this isn't necessary. There are other, far more important open source projects that do require donations, though.

MichaelMMS, 2022-02-17:
Thank you for the detailed answer. Yes, the communication with the device is strange. I wonder why there is no separate modern processor for TCP communication.  Then the manufacturer could just use a REST Api or WebSocket. Smart Home etc. is the future. I like the device, but the digital design could be better.

There are also a lot of bugs in the app. Data validation is also almost non-existent. For example, if you give the device a name with öÄÜ, the display is shot.I have already reported some bugs to the manufacturer, but this is probably not an issue there. Probably the firmware/operating system is programmed in C or Assember, so that input data is hardly validated. 
I have another device on stock, which I could connect to develop. Could also open the port...

What I just remembered, we are always looking for good developers, so if you are ever looking for a new job (>120k/y) just get in touch. Many greetings.

42Bastian, 2022-02-18:
"firmware/operating system is programmed in C or Assember, so that input data is hardly validated."

Validating of input data is by nothing related to the language software is written in. But yes, from what I see when looking _into_ the firmware it is very likely written in C.

closed 2022-05-05 by svalouch

svalouch, 2022-05-05:
Added a section to the documentation about write support.

do-gooder, 2024-08-02:
Would it be possible to implement and test a few important write operations? For example, for charging from the grid to be able to use Tibber?
power_mng.soc_strategy -> 2
power_mng.battery_power_extern -> -6000

commit ffbc7065 (2022-05-05): docs: document why there is no write-support in the cli

Fixes: #15