svalouch/python-rctclient #20: Neue Logging Events

hhoefling, 2022-09-23:
Ich habe durch vergleich mit der RCT App die folgenden Events indentifiziert

0x00000000  "TRAP" | time | time | unk | unk
0x00000017  "POWER BOATD"  | time | time | unk | unk
0x00000063  "PHASE_POS"  | time | time | unk | unk
0x0000007E  "ISLAND DETECTED"  | time | time | unk | unk

svalouch, 2023-10-05:
Hi, das ist sehr interessant! Existieren davon zufaellig tcpdumps (und ein Screenshot mit aufgeklappten Events) die ich analysieren koennte? Grade fuer die Checksummen waere das interessant.

Unabhaengig davon glaube ich hier ein System zu erkennen:
- `TRAP`: Bit 0 bei den fault codes heisst spannenderweise "TRAP occurred".
- `POWER BOARD`: Bit 23 heisst "Power Board Error".
- `PHASE_POS` habe ich unter `0x61` schon in der Tabelle, aber der zweite Wert wurde nicht als Timestamp angezeigt. Vermutlich ein Uebertraungsfehler meinerseits, denn Bit 99 bei den fault codes heisst "Phase position error".
- `ISLAND DETECTED` ist Bit 126 "Island detected".

Ich habe aktuell im Log `BAT_OVERVOLTAGE`, `BAT_OVERCURRENT` und `CAN_TIMEOUT` anstehen, und auch hier matchen alle perfekt mit den Fault codes! Das macht Sinn, denn so muss man nur die gesetzten bits (bei mehreren gleichzeitig anstehenden codes) isoliert betrachten und in einem array von Namen nachschlagen, um ein Display wie in der App zu erhalten.

Unabhaengig davon habe ich vor kurzem drei weitere Fault-Codes gefunden, von denen zwei nicht von `fault[*].flt` dargestellt werden koennen (bits 129+130, aber es gibt nur 128).

hhoefling, 2023-10-05:
Die gab es..... ist aber schon ein Jahr her....
Also nein.

sfudeus, 2023-10-07:
Oh, bin beim lesen auch über ein paar fehlende codes gestolpert. Mein aktueller Event-Output ist
```
0x10,2023-08-24T13:19:14,0x0,0x4248640b,0x0
0x10,2023-08-24T13:18:50,0x0,0x4248640b,0x0
0x57,2023-08-22T21:37:08,0x0,0x4321af48,0x0
0x63,2023-08-22T21:36:57,0x0,0x3ed5ec79,0x3e8a3d73
0xb,2023-08-16T11:47:48,0x0,0x40094459,0x3ddf3b6a
0x63,2023-08-12T21:24:00,0x0,0x40094459,0x3ddf3b6a
0x73,2023-08-12T20:46:56,0x3390cc2f,0x42f00000,0x43960000
0x36,2023-08-12T20:35:13,0x64d7d0e5,0x422bc53b,0x0
```

Dabei fehlen 0x10, 0x63, 0x36, 0xb.

[image]
[image]

tcpdump kann ich bei Bedarf noch ziehen.

sfudeus, 2023-10-07:
```
Frame complete: <ReceiveFrame(cmd=LONG_RESPONSE, id=6f3876bc, address=0, data=6521f0ff0000001064e73cb264e73cbc4248640b000000000000001064e73cb2000000004248640b000000000000001064e73c9a64e73ca44248640b000000000000001064e73c9a000000004248640b000000000000004f64e50e6464e50fb24329bcfd000000000000005264e50e6464e50fb24248640b000000000000005364e50e6464e50fb243362507000000000000005764e50e6464e50fb24321af48000000000000006364e50e5964e50e683ed5ec793e8a3d730000004f64e50e64000000004329bcfd000000000000005264e50e64000000004248640b000000000000005364e50e640000000043362507000000000000005764e50e64000000004321af48000000000000007964e50e5964e50e63417c055c3e428f6a0000007964e50e5900000000417c055c3e428f6a0000006364e50e59000000003ed5ec793e8a3d730000000b64dc9b4464dc9b4e400944593ddf3b6a0000000b64dc9b4400000000400944593ddf3b6a0000007964d7dc5064d7dc54404619bc3c9ba5e40000006364d7dc5064d7dc54400944593ddf3b6a0000007964d7dc5000000000404619bc3c9ba5e40000006364d7dc5000000000400944593ddf3b6a0000007364d7d3a03390cc2f42f00000439600000000003664d7d0e164d7d0e5422bc53b00000000)>
Received reply : logger.error_log_time_stamp              type: EVENT_TABLE       value: (datetime.datetime(2023, 8, 12, 20, 35, 13), {datetime.datetime(2023, 8, 24, 13, 19, 14): <EventEntry(type=0x10, ts=2023-08-24 13:19:14)>, datetime.datetime(2023, 8, 24, 13, 18, 50): <EventEntry(type=0x10, ts=2023-08-24 13:18:50)>, datetime.datetime(2023, 8, 22, 21, 37, 8): <EventEntry(type=0x57, ts=2023-08-22 21:37:08)>, datetime.datetime(2023, 8, 22, 21, 36, 57): <EventEntry(type=0x63, ts=2023-08-22 21:36:57)>, datetime.datetime(2023, 8, 16, 11, 47, 48): <EventEntry(type=0xb, ts=2023-08-16 11:47:48)>, datetime.datetime(2023, 8, 12, 21, 24): <EventEntry(type=0x63, ts=2023-08-12 21:24:00)>, datetime.datetime(2023, 8, 12, 20, 46, 56): <EventEntry(type=0x73, ts=2023-08-12 20:46:56)>, datetime.datetime(2023, 8, 12, 20, 35, 13): <EventEntry(type=0x36, ts=2023-08-12 20:35:13)>})
```