svalouch/python-rctclient #22: Cleared energies in RCT Inverter 6.0

parkmaj, 2023-01-13:
Hello,

First I would like to say thank you for your work.

I have used rctclient for about 2 years and it works great. But during that period accumulated energy inside RCT inverter was cleared two times. 

That means: 
* on display of RCT inverter total energy was shown as 0kWh
* energy.e_ac_total read through rctclient shows 0

Everything else is working correctly... also energy.e_ac_total starts again acumulate the energy (from 0)...

----

I would like to ask about your opinion. Could this behavior be casued by rctclient? Or it is more likely broken RCT inverter?

Have a nice day
Jan

weltenwort, 2023-01-13:
A related experience: I know of a Power Storage 10.0 system for which the `battery.used_energy` value has jumped to an unreasonably high value and doesn't change anymore.

To me this sounds like a memory hardware problem. For example the eeprom or flash might not be able to survive the required amount of write cycles. Or the RAM is not using ECC and is therefore susceptible to bit flips due to heat or cosmic rays.

42Bastian, 2023-01-13:
I also have these ridiculous high values. I asked RCT about, but got no answer.

parkmaj, 2023-01-13:
My case is slightly different, it is like reset of acumulated energy.

svalouch, 2023-10-03:
The inverter-side implementation has some bugs, though most should be ironed out in recent releases. One of these was that concurrent requests (e.g. from rctclient and an app, or multiple apps) could be mangled together and result in interesting behaviour, such as activating the display test. I've only seen this happen on firmware released in or before 2020 or so (have to check if I still have notes about that).
I specifically recall receiving a `Read` frame from an app as payload in a `Response` frame in the rctclient. Such things reek of broken buffer handling. RCT also has an interesting approach to TCP, it seems. Anyway, later releases fixed most of the bugs, so make sure to update your firmware.

The `rctclient` program itself won't deliberately change anything, writing isn't implemented for reasons I've outlined in another issue (except for timeseries). If you use it as a library (e.g. as part of another program), then it's up to that program and the bug should be filed there.

---

That being said, I firmly believe that the observed behaviour is a bug in the firmware and the vendor should fix that, or at least document it and make such documents available so measures can be taken.

svalouch, 2023-10-03:
As for the unreasonably high or stuck accumulated counters: It's a while since I've browsed through the app, perhaps there's a menu to reset the statistics somewhere?

Also, and this is probably a very bad idea and you'd be on totally your own if you tried: Write a value to change the counter contents. Perhaps it works and starts updating again? For all I know, it could break in various ways, so… yea, RCT should really fix things.

svalouch, 2023-10-05:
I'll go ahead and close this issue; as mentioned above, there isn't really anything that can be fixed on the client side.

closed 2023-10-05 by svalouch