svalouch/python-rctclient #23: rctclient command fails when installed as described in README.md

Eddy-the-Git, 2023-02-02:
Currently, when I install python-rctclient via the way described in README.md:

```
pip install -U rctclient
pip install -U rctclient[cli]
```

the rctclient command does not work, probably because the click version is 8.1.3 and the python-rctclient version is v0.0.3 (which uses old 'autocompletion' instead of 'shell_complete').

```
# rctclient --version
Traceback (most recent call last):
  File "/usr/bin/rctclient", line 5, in <module>
    from rctclient.cli import cli
  File "/usr/lib/python3.10/site-packages/rctclient/cli.py", line 108, in <module>
    def read_value(ctx, port: int, host: str, id: Optional[str], name: Optional[str], verbose: bool) -> None:
  File "/usr/lib/python3.10/site-packages/click/decorators.py", line 308, in decorator
    _param_memo(f, OptionClass(param_decls, **option_attrs))
  File "/usr/lib/python3.10/site-packages/click/core.py", line 2495, in __init__
    super().__init__(param_decls, type=type, multiple=multiple, **attrs)
TypeError: Parameter.__init__() got an unexpected keyword argument 'autocompletion'
```

blueslow, 2023-02-14:
Yes

> the rctclient command does not work, probably because the click version is 8.1.3 and the python-rctclient version is v0.0.3 (which uses old 'autocompletion' instead of 'shell_complete').

Yes Eddy-the-git you seem right I downgraded click  to version 8.0.0 and the rctclient command  did not failed with "... unexpected keyword argument 'autocompletion'..."
I used the following command line code in my virtual environment as a workaround:
`pip uninstall click`
`pip install --force-reinstall -v "click==8.0.0"`

A better solution would of course  be to fix the code.

Gill-Bates, 2023-02-23:
So the `requirements.txt` should look like this:

```
click>=8.0
Sphinx>=2.0
sphinx-autodoc-typehints
sphinx-click
sphinx-rtd-theme
```

Gill-Bates, 2023-02-23:
> Yes Eddy-the-git you seem right I downgraded click to version 8.0.0 and the rctclient command did not failed with "... unexpected keyword argument 'autocompletion'..." I used the following command [...]

Instead of removing the door to make it stop squeaking, I would adjust the client to work with the current version of Click. :-)

Eddy-the-Git, 2023-02-23:
Actually, in the main branch here it was fixed 10 months ago, however `pip uninstall click` installs the tagged version v0.0.3, which does not contain the fix...

Eddy-the-Git, 2023-02-23:
... so the proper fix would be to upload the main branch to pip (and do some tests before that with recent versions of all the tools, because there may be other issues lurking). I am guessing only Stefan can do that.

poggenpower, 2023-03-19:
I can't confirm that an install from the repository is working:
`pip install git+https://github.com/svalouch/rctmon` 
will also install `click-8.1.3-py3-none-any.whl`  and will fail with the same error.

aruttkamp, 2023-08-02:
> ~I can't confirm that an install from the repository is working: `pip install git+https://github.com/svalouch/rctmon` will also install `click-8.1.3-py3-none-any.whl` and will fail with the same error.~
> 
> Sorry it is working if you use the right repo: `pip install git+https://github.com/svalouch/python-rctclient`

Works ! Thank you

svalouch, 2023-10-05:
I've pushed a tag (`v0.0.4`) and will publish that version on pypi soon. I'll also look into modernizing the setup side of things soon-ish (I've already fixed a bug with dependency declaration in this release), sorry for the long absence.

closed 2023-10-05 by svalouch