svalouch/python-rctclient #25: io_board: home-relais status

SaxonWood, 2023-11-08:
Für das "home_relais" alias Multifunktions-Relais auf dem IO-Board gibt es gemäß Registry ein paar OIDs, die die per App eingestellten Schwellwerte und den eingestellten Modus zeigen. Was mir aber unklar ist, ob man den aktuellen Relais Status auch abfragen kann, um z.B. zu sehen, ob der Wechselrichtern diese Relais angezogen hat.
Gibt es aus den Erfahrungen eine OID, die direkt oder per Bitmask diesen Status reflektiert? 
(Bei zielführenden Hinweisen auf einen solche OID würde ich das auch per Test nachprüfen hier das Ergebnis posten.)

alexander-stefan, 2024-04-30:
Dieses Thema finde ich auch spannend. Ich würde gerne abfragen ob das Multifunktions-Relais
- gerade aktiv ist
- wie oft es am aktuellen Tag geschaltet wurde

Gibt es außerdem die Möglichkeit das Multifunktionsrelais über die Schnittstelle zu schalten?

svalouch, 2024-04-30:
In der App sehe ich bei "Multifunctional relay usage" nur "Load" und "Alarm", das werden die einzigen Steuerungsmodi sein, beim "Digital I/O" sieht es aenlich aus mit den entsprechenden Modi. Ob es mehr gibt muesste man also beim Hersteller erfragen, auch wie man es anspricht.

Ob es schaltet sieht man mit grosser Wahrscheinlichkeit nur in der Event-Liste, wenn ueberhaupt (bei Verwendung als S0 in/output macht das naemlich ueberhaupt keinen Sinn das zu Event zu loggen). Generell kann man den Traffic mitschneiden (Wireshark/tcpdump) und mit `tools/read_pcap.py` versuchen zu dekodieren, und das Ergebniss dann manuell durchsuchen ob das gewuenschte dabei ist. Ansonsten auch hier wieder: Nur der Hersteller kann das abschliessend beantworten.

SaxonWood, 2026-01-11:
Das Kürzel "rse" im OID Namen weckt Erinnerungen an "Rundsteuerempfänger" für eine netzseitig verordneten Reduktion der Einspeiseleistung. Auch wenn RCT das meines Wissens nicht unterstützt und mir auch keine Anschlussmöglichkeit dafür präsent ist, sind die Ähnlichkeiten mit den Relais K1..K4 in der Tabelle auf Seite 3 des verlinkten Dokuments frappierend, speziell auch die 30% und 60% Reduktion. So als ob die OIDs `io_board.rse_table[n]` einen Lookup-Table für die Reduktionswerte in Abhängigkeit der Relaisstellungen K1..K4 sind.

[https://www.gw-energienetze.de/fileadmin/gw-energienetze/40_service/40.6_veroeffentlpflichten/2022-01-31_informationsblatt_rundsteuerempfaenger_stolzenau.pdf](url)