svalouch/python-rctclient #36: Update on "RCT Power Serial Communication  Protocol" ?

HWE70, 2024-09-06:
The latest version I know is version 1.14. I think there must be a new one for the software 2.3.5676, as this release finally include the support for the second battery tower (twin-tower/Doppelturm). For this reason I would expect additional IDs for the second battery tower. As far as I can the values populated by rctclient in a twin-tower setup are the values of the primary battery tower.

sfudeus, 2024-09-08:
The only obvious field in the app is "number of installed battery towers", changing this causes write events on OID 0xbd870886. Having a 2nd tower surely would help :-)

HWE70, 2025-05-04:
Hi, I just did some investigations. Most OIDs which are needed for the second battery are already defined and can be fetched with rctclient. I documented this here: https://github.com/weltenwort/home-assistant-rct-power-integration/issues/396

I found a few additional OIDs, which are still unknown (types match the length, but may be wrong):

```
OID[0xBD870886] = { name="new.001", type="UINT8" }
OID[0x99EA35E2] = { name="new.002", type="UINT8" }
OID[0x44D4CC2B] = { name="new.003", type="UINT32" }
OID[0x84627EAA] = { name="new.004", type="FLOAT" }
OID[0xE51B041C] = { name="new.005", type="FLOAT" }
OID[0x09648B47] = { name="new.006", type="FLOAT" }
```

However my basic request, the dual power support is fixed :-)

closed 2025-05-04 by HWE70