svalouch/python-rctclient #38: unpack error, that points to the "decode_value" function

Pythonaire, 2025-03-04:
Got continuously "failed: unpack requires a buffer of 4 bytes" error, that points me to an error inside the decode_value function.

sfudeus, 2025-03-05:
For any id you are requesting or for specific ones only? Do you have an example?
A tcpdump showing the packets would be helpful as well.

sfudeus, 2025-03-07:
In my setup I cannot reproduce. What kind of inverter do you have, with which firmware version?
And in addition to a tcpdump, at least the output of `rctclient --debug ` would be helpful while having an unpack error.

Pythonaire, 2025-03-08:
Hi Stephan,

Many thx.

I read the following id's continously in a 5 minutes interval from a RTC6000 inverter:
```
dc_conv.dc_conv_struct[0].p_dc_lp
dc_conv.dc_conv_struct[1].p_dc_lp
energy.e_dc_year[0]
energy.e_dc_year[1]
g_sync.p_ac_grid_sum_lp
g_sync.p_ac_load_sum_lp
energy.e_load_year
energy.e_grid_load_year
energy.e_grid_feed_year
g_sync.p_acc_lp
```
With the cli debug mode, i cannot reproduce the error. 

In my produced logs, i see in the last 8 hours (approximately one message per hour.)

```
[Error] Attempt 1 reading dc_conv.dc_conv_struct[0].p_dc_lp failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading energy.e_grid_feed_year failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading dc_conv.dc_conv_struct[0].p_dc_lp failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading dc_conv.dc_conv_struct[0].p_dc_lp failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading energy.e_load_year failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading dc_conv.dc_conv_struct[0].p_dc_lp failed: unpack requires a buffer of 4 bytes
[Error] Attempt 1 reading dc_conv.dc_conv_struct[0].p_dc_lp failed: unpack requires a buffer of 4 bytes
```

On second attempt (one seconds later) it works. So it's not really a problem, but im curious, what's happing here.

The error might point to the id dc_conv.dc_conv_struct[0].p_dc_lp, due to the frequency. I rewrote my script to write a tcp dump file. I will load that up here, if the error occurs again.
Many thx for your support!

Peter

Pythonaire, 2025-03-09:
At least, the rapidly access to the inverter data seems to be the reason of the "unpack" error. Since i set a time delay between each get_value() call, no errors. 
The errors only comes up on permanently written ids by the inverter itself, for example "dc_conv.dc_conv_struct[0].p_dc_lp". Reading ids like "energy.e_dc_day[0]" at midnight in the same way, no errors.

closed 2025-03-09 by Pythonaire

weltenwort, 2025-03-12:
> The interesting fact is, i used your repository over years on a raspberry pi without any errors, Now i moved to a bolder x86 CPU machine and faced these errors.

The firmware updates that RCT Power releases also seem to sometimes change the behavior of the API. So maybe there just is a random coincidence with your host hardware change.